iqm-finland/KQCircuits #3: Format license file for metadata

nathanshammah, 2021-09-22:
I understand that the license is just GNU GPL 3.0. However, on the homepage, this fact is not automatically picked up by Github. I propose to use Github standard template (which can be modified with company details on copyright) so that this information is displayed. Example: https://github.com/unitaryfund/mitiq/blob/master/LICENSE. I could open a PR about it.

jheinsoo, 2021-09-22:
Thank you for reporting, we will consider it! Changes to license, even formatting, will be always nontrivial as you likely understand.

jkotilahti, 2021-09-22:
Thank you for reporting this @nathanshammah! It seems the problem is actually not with the LICENSE file, since it has identical text to the GitHub template. The problem is due to other files in project root with "license" or "copyright" in their name, which will confuse the license detector used by GitHub, see https://licensee.github.io/licensee/what-we-look-at/. This can be fixed by renaming or moving `COPYRIGHT`, `license_manifest.txt` and `check_copyright_headers.py`, working on it now.

jkotilahti, 2021-09-23:
Fixed by b866a05.

closed 2021-09-23 by jkotilahti