iqm-finland/KQCircuits #6: How to generate a new element

pauljametsuria, 2022-02-07:
Hello,

I created a new element in the qubit library named Coupler. To create it I modified the code swissmon.py and I can use it in the Klayout GUI as any other element: 

[image]

I would like to generate my new element using the macro environment as Coupler.create(layout, .... ) as I can do for any other element but I get this error message in the macro environment:  

[image]

How can I set this up so I can generate my element from macro? Do I need to  add my element in some element list inside the code somewhere? Some documentation on how to create new elements and integrate them in the package would be greatly appreciated.

Thank you,

Best, 

Paul

jkotilahti, 2022-02-07:
Thank you for reporting the issue here @pauljametsuria! This error happens if you have not imported `Coupler` in the macro where you are using it. Assuming that you added `Coupler` in KQCircuits `qubits` folder, then you should add the line 
`from kqcircuits.qubits.coupler import Coupler` at the top of the macro to make it work. Let me know if this helps, or if there is some other problem.

I agree that better documentation for creating new elements and integrating them in the package would be useful, we are planning to add it to https://iqm-finland.github.io/KQCircuits/user_guide/python_workflow.html#example-of-defining-a-chip-and-an-element.

pauljametsuria, 2022-02-07:
That worked :) 

Thank you!

jkotilahti, 2022-02-08:
Good! I'll close this issue now since the problem was solved. We will also be improving the documentation in the near future.

closed 2022-02-08 by jkotilahti