iqm-finland/KQCircuits #9: Links to files in documentation

nikosavola, 2022-03-31:
The docs occasionally refer to specific files in repository, such as, this paragraph in [Geometry from Klayout GUI](https://iqm-finland.github.io/KQCircuits/user_guide/export/simulation_export.html#geometry-from-klayout-gui):

> Alternatively, run following macros in Klayout:
>* `scripts/macros/export/export_ansys.lym`
>* `scripts/macros/export/export_sonnet.lym`

It would be nice if these were links to the files in `main` on GitHub. The `|GIT_CLONE_URL|` substitution [already exists](https://github.com/iqm-finland/KQCircuits/blob/09ef6c7198fb88c4084b4dbb13948e755b4b8c4a/docs/conf.py#L174), so it can probably be used to implement this.

divshacker, 2022-06-02:
I would like to work on this issue. Please Assign me 
Thanks

jkotilahti, 2022-06-02:
Thank you for offering to help with this @divshacker! I assigned you now.

Please note that in order for us to accept your pull requests you must sign the IQM Contributor License Agreement (https://github.com/iqm-finland/KQCircuits#contributing).

nikosavola, 2022-06-03:
It would be nice if this was done automatically for strings that look like paths, or with some custom Sphinx directive like 
```
:github-link:`scripts/macros/export/export_ansys.lym`
```
The GitHub repo configurations exists already `conf.py`. So this wouldn't need hardcoding any repo URL either.