iqm-finland/KQCircuits #13: Macro for generating a list of refpoints from all hierarchy levels

jkotilahti, 2022-05-11:
To help users get a better understanding of refpoints, we could have a KLayout macro that prints out a list of all refpoints in the layout. The output should be formatted so that the hierarchy of the refpoints/cells is clear. See https://iqm-finland.github.io/KQCircuits/user_guide/python_workflow.html#refpoints for an explanation about refpoints.

divshacker, 2022-06-16:
Hey, I am interested in working on this issue. May I know What is KLayout macro and how it prints list?
Thanks

jkotilahti, 2022-06-16:
Hi @divshacker! You can find information about KLayout macros in KQCircuits documentation https://iqm-finland.github.io/KQCircuits/user_guide/macro_workflow.html and in KLayout documentation https://www.klayout.de/doc-qt5/about/macro_editor.html.

In KQCircuits we have for example the `selection_length` macro, which prints out lengths of waveguides to the macro editor console. If you have installed KQCircuits in KLayout, then you can find this macro in `kqcircuits_scripts/macros/measure` folder in `Python` tab of the KLayout macro editor window. If you drag-and-drop a WaveguideCoplanar element into the layout and select it, then running this macro will print out its length.

LeonWubben, 2024-01-02:
The mentioned macro has now been adjusted to show the hierarchy. Including an option to only print the refpoints whose names or cell match a given regex and an option to only print those refpoints that are within a certain range of coordinates.

closed 2024-01-02 by LeonWubben