iqm-finland/KQCircuits #14: Manual steps in code refresh

jkotilahti, 2022-05-11:
If I edit a PCell source code and want to see the changes within KLayout application, I need to:
1. run [Reload libraries](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/macros/reload.lym) macro
2. make a new copy of a PCell

Would be nice if there was one less manual step and I would not need to make a new PCell. Instead:
1. before reloading, the reload macro could extract the PCell information in the active layout 
2. removes the PCell instance
3. reloads the libraries
4. places instances of the PCells with same parameters and location

Maybe after this issue we should have the follow up features:

5. watchdog for code changes to autorefresh such that I would not even need to run refresh macro manually
6. partial refresh to refresh a single python module and corresponding PCell, not the full library (the changed PCell might be inside other PCells)

qpavsmi, 2023-05-03:
# Some further clarification for UnitaryHack participants

## Prerequisites

- Requires [GUI](https://iqm-finland.github.io/KQCircuits/developer/setup.html) KQCircuits installation.

## Use case example

Open KLayout (in edit mode) and drag-and-drop, lets say, a Swissmon qubit into the layout. You will find the Swissmon qubit entry from bottom left `Libraries` panel by choosing the `Qubit library - Library for qubits.` from the dropdown. With the swissmon qubit placed somewhere in the layout, without closing the KLayout window, edit the [swissmon.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/qubits/swissmon.py) code, for example, by decreasing iterations to the for loop that calls `self._produce_coupler` to have `range(2)`. Then in KLayout window, from the top toolbar choose `KQCircuits > Reload Libraries`.

[image]

Seemingly, there is no difference on the Swissmon already placed in the layout. However, by double-clicking the Swissmon you will notice that the PCell parameters can no longer be changed. Now drag-and-drop another instance of Swissmon. Now you will see that the Swissmon placed after pressing `Reload Libraries` has a right-hand coupler missing, which is caused by our edit in [swissmon.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/qubits/swissmon.py) code. You can also verify that editing PCell parameters for the newer Swissmon is possible.

[image]

This should be changed so that after pressing `Reload Libraries` the old instance of the Swissmon qubit would get updated according to the current state of the code. The PCell parameters should also be available for editing. This would involve removing PCells and memorizing their coordinates, classes and PCell parameters, then reloading the library and then placing the PCells back to their coordinates with PCell parameters intact (just as outlined in above comment).

## How to get started

When `Reload Libraries` is pressed, it runs a macro [0reload.lym](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/macros/0system/0reload.lym). Notice that macros are a python script wrapped in some xml structure. It could be more convenient to edit macros from KLayout's own macro editor (from top toolbar: `Macros > Macro Development (F5)`).

[image]
[image]

Right now [0reload.lym](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/macros/0system/0reload.lym) only calls `load_libraries` from [library_helper.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/util/library_helper.py) which does the library reloading. This macro should be extended to include the extra features.

## Implementation tips

Following are some of the methods from [KLayout](https://www.klayout.de/doc-qt5/code/) and [KQCircuits](https://iqm-finland.github.io/KQCircuits/api/kqcircuits.html) API to help you implement this feature. We do not guarantee that the feature gets implemented using only these methods.

You will need to reconstruct every PCell placed on every open layout of a KLayout window where the `Reload Libraries` was pressed.

[image]

To iterate over each [LayoutView](https://www.klayout.de/doc-qt5/code/class_LayoutView.html) use `pya.Application.instance().main_window()` to access the [view](https://www.klayout.de/doc-qt5/code/class_MainWindow.html#method175) method that returns a [LayoutView](https://www.klayout.de/doc-qt5/code/class_LayoutView.html) by integer index. The number of available views is returned by the [views](https://www.klayout.de/doc-qt5/code/class_MainWindow.html#method176) method. Notice that `pya` is a built-in module in the KLayout macro environment. Then for each [LayoutView](https://www.klayout.de/doc-qt5/code/class_LayoutView.html) object use the [active_cellview](https://www.klayout.de/doc-qt5/code/class_LayoutView.html#method24) to get a [CellView](https://www.klayout.de/doc-qt5/code/class_CellView.html) object and then [cell](https://www.klayout.de/doc-qt5/code/class_CellView.html#method10) to access the current top cell of that view`[*]`. Then iterate over every instance under the top cell with [each_inst](https://www.klayout.de/doc-qt5/code/class_Cell.html#method51).

To retrieve the location of an instance use [trans](https://www.klayout.de/doc-qt5/code/class_Instance.html#method86) method to get a [Trans](https://www.klayout.de/doc-qt5/code/class_Trans.html) object. This object contains the full transformation of the instance, including position, rotation, scale and mirroring. The instance can be converted into a [PCellDeclaration](https://www.klayout.de/doc-qt5/code/class_PCellDeclaration.html) object using the [pcell_declaration](https://www.klayout.de/doc-qt5/code/class_Instance.html#method75) method. If the method returns `None`, then the instance was not a PCell and can be skipped. Use the python built-in method `__class__` to access the implementing class of the `PCellDeclaration`, which would be [Swissmon](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/qubits/swissmon.py) in our example use case. Then cast the `PCellDeclaration` object into [Element](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/elements/element.py) and use `pcell_params_by_name` to retrieve PCell parameters of the instance.

[... cut ...]
- Pressing `Reload Libraries` should reload all PCells in all open layouts of the KLayout window so that the PCells are rendered according to most up-to-date code, without the need of creating another PCell after the fact
- No tests needed

## More features

As outlined in above comment, it would be nice to have a mechanism that notices the change in the source files which would prompt KLayout to automatically perform library reload. We will leave this as a future task. You are welcome to submit an additional PR after unitaryhack if you are interested in implementing this.

iqmtestd, 2023-09-14:
It is a bit undefined what should happen if a default parameter value also changes in the code. We could either use the instance's parameter value or the new default value.  In this case it is probably better to stick with the instance's value. Unless, perhaps, if the instance was using the default value then after reload it could use the new default value.

commit 9fc605e4 (2023-09-25): Reload existing PCells without closing KLayout

The "KQCircuits -> Reload libraries" command apart from reloading the
libraries also saves and re-creates all PCells in every view using
the updated code but retaining all the parameter changes of the user.

Fix #14
commit 5de2b203 (2023-09-28): Reload existing PCells without closing KLayout

The "KQCircuits -> Reload libraries" command apart from reloading the
libraries also saves and re-creates all PCells in every view using
the updated code but retaining all the parameter changes of the user.

Fix #14
PR #62 (2023-09-28): Reload existing PCells without closing KLayout
The "KQCircuits -> Reload libraries" command apart from reloading the libraries also saves and re-creates all PCells in every view using the updated code but retaining all the parameter changes of the user.

Fix #14
closed 2023-09-28 by iqmtestd