iqm-finland/KQCircuits #21: Backward compatibility

srjmas, 2022-11-15:
Recently I updated the kqc version to 4.5.0, and some of my older designs wouldn't open anymore. For example, in the same mask, chips of different time are opened differently:

[image]

This is the error I see when zooming in:

[image]

qpavsmi, 2022-11-15:
Seems that this issue in particular is caused by our change in the face naming system. We used to have 'b' to denote the face of the bottom chip and 't' to denote the face of the top chip. We have now generalized the face naming convention to support arbitrary number of chips. Now 'b' face is renamed as '1t1', which stands for 1st chip, top face, 1st layer. 't' face is renames as '2b1', which stands for 2nd chip, bottom face, 1st layer.

[image]

I believe that to fix this issue, you'll simply need to change the face_ids ("Chip face IDs list") PCell parameter for your chips from 'b,t,c' to '1t1, 2b1, 2t1'.

As for the second question, while we do our best to make our development as backwards compatible as possible, certain changes in KQC will cause older designs to break, and it's good practice to maintain the designs as KQC gets updated. We will always have the older versions of KQC available, so these designs won't get lost.

I also recommend to save a second version of your designs without PCells. If you're saving your designs using 'Save As', in the prompt uncheck the 'Store PCell and library context information' like so:

[image]

This will save your design simply as geometry data, which means nothing in KQC could break it, even a KLayout application without KQC will open it just fine. The drawback is that parameters for PCells will not be stored if the design is saved that way.

Let me know if this fixes your issue and thank you for your interest in KQCircuits.

srjmas, 2022-11-15:
Thank you, it fixed most of the issues. The launchers disappeared
completely though - i am not able to access their properties, it is like
they have zero size.
[image: image.png]
It looks like I have to go to every instance and click on them
individually, as I am not using a KQC chip element as top cell. Is there
some way to go through all the PCells in the layout automatically and
modify the faces parameter?

qpavsmi, 2022-11-15:
The image unfortunately is not displaying to me

For the immediate, you could write a macro to do this. Open Macros > Macro Development (F5), create a new plain Python file, then something like the following code should change the face_ids value for each selected PCell
```
from kqcircuits.pya_resolver import pya
from kqcircuits.klayout_view import KLayoutView
from kqcircuits.elements import Element

view = KLayoutView(current=True)
layout = KLayoutView.get_active_layout()
layout_view = KLayoutView.get_active_cell_view().view()

for obj in layout_view.object_selection:
    if obj.is_cell_inst():
        cell = obj.layout().cell(obj.inst().cell_index)
        if cell.pcell_declaration() is None:
            continue
        if not issubclass(cell, Element):
            continue
        cell.face_ids = ["1t1", "2b1", "2t1"]
```
(Haven't tested this code myself, apologies for that.) Then choose each element in the layout for which you wish to change the face_ids value and click "Run script from the current tab"

For the long term, to have a development environment that is more robust to updates in KQC, I recommend installing KQC according to this guide: https://iqm-finland.github.io/KQCircuits/developer/setup.html . Then you can construct the chip using python code, there are some examples of how we do it in KQCircuits code in `chips` directory.

qpavsmi, 2022-11-15:
I apologize for providing a non-working macro, it indeed does not work at all. I have written a newer version of the macro that I have succesfully run on your provided gds file. It goes as follows:

```
from kqcircuits.pya_resolver import pya
from kqcircuits.klayout_view import KLayoutView
from kqcircuits.elements.element import Element

view = KLayoutView(current=True)
layout = KLayoutView.get_active_layout()

cells = layout.top_cells()
for cell in cells:
    iter = cell.begin_instances_rec()
    while not iter.at_end():
        instance = iter.current_inst_element().inst()
        if instance.pcell_declaration() is not None and instance.pcell_parameter("face_ids") is not None:
            instance.change_pcell_parameter("face_ids", ['1t1', '2b1', '2t1'])
        iter.next()
```
No selection on the layout is needed, it will change the "face_ids" for all cells placed on the layout recursively.
There is some quirk after running the macro that it will create duplicate cells on the same hierarchy level as "Saxophone" cell. If you right click on one of such cells and choose "Delete cell", all other duplicated cells will mysteriously disappear, with only the correct cell remaining. Have to admit I don't have an explanation for this. Please review carefully that your cell remains intact after running this macro and is according to your design.

Find attached a fixed version of "Saxophone" that I have produced on my machine using this macro. Please let me know if there are any more issues for any of your designs.

[Saxophone_fixed.zip](https://github.com/iqm-finland/KQCircuits/files/10016380/Saxophone_fixed.zip)

srjmas, 2022-11-15:
Amazing, all worked smoothly on 3 more chips from various times. Thank you very much for your quick and effective help!

closed 2022-11-15 by srjmas