iqm-finland/KQCircuits #24: Installing missing python packages in windows

srjmas, 2022-11-18:
Failed installation of KQC 4.5.0 in Klayout 0.27.2 and 0.26.12 on Windows 10

First a dialog appears:
---------------------------
Install packages?
---------------------------
Install missing packages using 'pip': autologging, tqdm, networkx
---------------------------
&Yes   &No   
---------------------------

I click Yes, no visible response
Then startup macros are executed with the following errors:

C:\Users\srj\KLayout\salt\KQCircuits\python\kqcircuits\util\library_helper.py:33: No module named 'autologging'
  C:\Users\srj\KLayout\salt\KQCircuits\python\kqcircuits\util\library_helper.py:33
  C:/Users/srj/KLayout/salt/KQCircuits/python/scripts/macros/library.lym:20 (class ModuleNotFoundError)
C:\Users\srj\KLayout\salt\KQCircuits\python\kqcircuits\klayout_view.py:19: No module named 'autologging'
  C:\Users\srj\KLayout\salt\KQCircuits\python\kqcircuits\klayout_view.py:19
  C:\Users\srj\KLayout\salt\KQCircuits\python\kqcircuits\util\macro_prepare.py:20
  C:/Users/srj/KLayout/salt/KQCircuits/python/scripts/macros/startup.lym:19 (class ModuleNotFoundError)

Trying 'pip' in command prompt leads to 'is not recognized'

What can be done?

qpavsmi, 2022-11-21:
Did you install with a Salt package or by following this guide: https://iqm-finland.github.io/KQCircuits/developer/setup.html ? I'd recommend reinstalling KQC and to try again, unless you have already tried that. I would also recommend using a newer KLayout version, like 0.27.11 (newest version is 0.27.12 but we have recently found some of the KQC features not working for it and are working on fixing those)

srjmas, 2022-11-21:
Thank you, it seems to work now!

I think that the Salt package remained alive through several reinstallations, until I started to delete folders manually. But after it was gone (no KQC libraries after fresh KLayout re-install), the developer guide worked just fine with Klayout 0.27.11

closed 2022-11-21 by srjmas