iqm-finland/KQCircuits #45: Add example Fluxonium

nikosavola, 2023-03-08:
### Description

At APS March meeting 2023 there was large interest in using KQCircuits for Fluxoniums designs. Would be nice to have one example qubit with a Junction array. 

### Use case

Use KQC for designing Fluxonium devices.

nikosavola, 2023-03-13:
I am happy to help if anyone is interested in adding a superinductor array to, for example, Double Pads.

qpavsmi, 2023-04-26:
# Issue specification for UnitaryHack participants

## Prerequisites

KQCircuits [GUI installation](https://iqm-finland.github.io/KQCircuits/developer/setup.html) required

## Issue description

The task is to implement a simple version of the Fluxonium qubit into the KQCircuits library. The geometry of the qubit should be parametrised so that certain features in the qubit could be tuned on-the-fly without rewriting the code.

## How to get started

We recommend getting started by copying an existing qubit in KQCircuits and using that as a template for the new Fluxonium qubit. Let's copy double_pads.py into the same folder `qubits` and rename it `fluxonium.py`. Rename the name of the class in the python file from `DoublePads` to `Fluxonium`.

Now when you start KLayout, change the Libraries selection to `Qubit Library - Library for qubits.`. You should see the new `Fluxonium` entry in the library list. Drag-and-drop that entry into the layout, then press `F2` and `*` to center the view and unhide the cells. You should be able to see geometry for the double pads qubit, but under the Fluxonium name.

Let's get back to editing `klayout_package/python/kqcircuits/qubits/fluxonium.py`. Some main take-aways from the previous code:
- The `Fluxonium` (previously `DoublePads`) class is a subclass of the `Qubit` class.
- All of the geometry generation happens under the `build` function. All other methods present in the code are private helper functions.
- In the header of the class we define parameters for the qubit, e.g. `ground_gap_r = Param(pdt.TypeDouble, "Ground gap rounding radius", 50, unit="μm")`. We only have `unit` argument for scalar values. These parameters can then be used with `self.ground_gap_r`. If you define a `pdt.TypeList` type parameter, make sure to cast all elements to appropriate datatypes because they are interpreted as strings when changed from GUI. (e.g. `float(self.ground_gap[0])`).
- To produce geometry, first define a shape by using pya.DPolygon or pya.DBox. Then convert the shape into a pya.Region with `region = pya.Region(poly.to_itype(self.layout.dbu))`, where `poly` is the shape. If you have multiple regions that possibly overlap, you can combine them into a single region using the `+` operator: `region = region1 + region2`. Finally, add the region into the qubit cell `"SIS_junction"` layer with `self.cell.shapes(self.get_layer("SIS_junction")).insert(region)`.
- Ports are used as reference points to connect components together within the chip. Add a port using `self.add_port("port_name", port_origin, direction=port_direction)`, where `port_origin` is pya.DPoint and `port_direction` is pya.DVector. Direction helps to align a waveguide into the port. May not be needed.
- If there are parameters needed from other Element classes, you can add those to the `Fluxonium` qubit with `@add_parameters_from(...)`. This will probably not be needed.

For more information refer to the [KLayout API](https://www.klayout.de/doc-qt5/code/) and [KQCircuits workflow advisory](https://iqm-finland.github.io/KQCircuits/user_guide/python_workflow.html).

To see the changes in the code, from KLayout top task bar choose `KQCircuits > Reload libraries`, then drag another copy of the `Fluxonium` qubit into the layout. If there are errors in the code, it will show you the stacktrace on the layout.

You can also open your qubit from IDE as explained [here](https://iqm-finland.github.io/KQCircuits/user_guide/python_workflow.html).

## Fluxonium design

The outline of the design is presented in this chart which we will further describe in text.

[image]

Fluxonium is an array of junctions (blue lines) connected by wires (red lines). Junctions should align to y-axis and wires to x-axis, and they should be orthogonal to each other. Sequence of junctions should form a line, with wires connecting them in the middle. Should the junction sequence make a bend, the wires can be placed somewhere not strictly in the middle - as long as they connect the two junctions. Junctions and wires should be placed in the `"SIS_junction"` layer.

On the bottom, interfacing the antenna, there should be two wires connecting two junctions to form a SQUID loop area. Multiple SQUID loop areas should be in a sequence as well. From each end of the SQUID array the wire should lead to a wider antenna.

We recommend to think of the Fluxonium construction as a set of line segments, where each line segment is populated by junctions along the line. If the line is diagonal, the wires should remain to be aligned with the x-axis and form a shape of stairs as they connect the junctions. On the sides of the Fluxonium there should be one junction enabling a U-turn of the junction sequence. See the below chart for line segment visualisation.

[image]

The fabrication method that will most likely be used for this design is so-called undercutting, for which we need shapes that are placed in the `"SIS_juntion_2"` layer. These shapes are visualised in the chart as pink rectangles, above and below the wires, as well as pertruding above and below both ends of junctions. For wires, the undercutting regions should alternate between being above or below the wire. Consult the chart if the direction of the undercutting region is unclear in specific cases. Open the GDS file within attached ZIP file with KLayout for further reference (courtesy of Vladimir Milchakov). Here layer `1/0` would be `SIS_junction` and `2/0` would be `SIS_junction_2`.

[... cut ...] ification for them not being parameterised.

## Definition of done

- When Fluxonium qubit is drag-and-dropped into the layout, it should show required design with no errors or failures. Same if it is run with `create_element_from_path.py`.
- Parameters can be changed from KLayout UI without it causing errors (if using reasonable values).
- Code should have sufficient docs: `Fluxonium` class should have a clear explanation, parameters should be well described.
- No tests needed for qubit element

hrahman12, 2023-06-01:
Hi I'm interested in working on this

qpavsmi, 2023-06-01:
> Hi I'm interested in working on this

Awesome, thank you for your interest! You can start working on the solution and once the solution is ready, create a pull request. During review we might give some suggestions to refine the solution, but once it is ready, we will assign you to this issue and mark it closed. Don't hesitate to make a pull request earlier rather than later so that we could have some time before the deadline to reiterate your solution from our feedback if needed.

Sarah-Dweik, 2024-06-21:
Hello;
I am facing this issue when I try to add the super inductor junction on a Double pad qubit, I wonder what I can do about it.

[image]
[image]

qpavsmi, 2025-06-13:
Issue completed, see https://github.com/iqm-finland/KQCircuits/commit/3a836d89f6f302f4c704b68cf47853c868aa7f9e and https://github.com/iqm-finland/KQCircuits/commit/6179773f094a0a331c8d35c6125222cdb38db9b4