iqm-finland/KQCircuits #48: Provide Vim command for `create_element_from_path.py`

nikosavola, 2023-04-11:
### Description

The [`create_element_from_path.py` macro](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py) currently has info on setting it up with VSCode and PyCharm. I think adding an example command for vim/neovim makes sense due to their popularity. The example should work platform-independently and be easy to add a `.vimrc` config.

### Use case

A new `:command` is provided in Vim along with a macro for running it. This would generate the current file in KLayout GUI.

qpavsmi, 2023-05-02:
# Issue specification for UnitaryHack participants

## Prerequisites

- Requires [GUI](https://iqm-finland.github.io/KQCircuits/developer/setup.html) KQCircuits installation.

## Issue description

We currently include a utility script with the KQCircuits source code called [util/create_element_from_path.py](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py). If an element is given as an argument, let's say [kqcircuits/elements/smooth_capacitor.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/elements/smooth_capacitor.py), the script will open a KLayout UI window with an instance of the given element placed in the layout. The script is useful for element development workflow to quickly see the results of the code in KLayout. For this reason we want to integrate this script into as many IDEs as we can. We have currently outlined instructions on how to set up this script as an external tool in [PyCharm](https://www.jetbrains.com/pycharm/) and [VSCode](https://code.visualstudio.com/), instructions are in the comment section of [util/create_element_from_path.py](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py). The task is to write similar instructions to have the script invoked from the [vim](https://www.vim.org/) and [neovim](https://neovim.io/) command line.

## How to get started

Try running the following from command line:
`klayout -e -rm util/create_element_from_path.py -rd element_path=kqcircuits/elements/smooth_capacitor.py`
This uses KLayout to invoke [util/create_element_from_path.py](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py), in this example to show [klayout_package/python/kqcircuits/elements/smooth_capacitor.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/elements/smooth_capacitor.py) element. `klayout` might not be recognized as a command. The reason is that depending on OS the executable for KLayout is placed at a different path. For Linux platforms the path should be `/usr/bin/klayout` (and should be a recognized application). For Windows the path is `%APPDATA%\KLayout\klayout_app.exe`. For Mac it is either `${USER}/Applications/KLayout/klayout.app` (if installed from HomeBrew) or `${USER}/Applications/klayout.app`. For now, either set up an alias for `klayout` to point to the executable in your system or replace `klayout` with the path manually.

The command should cause KLayout window with a smooth capacitor on it. You can also try other values for `element_path` argument, as long as it's an element defined in KQCircuits repository (under directories `elements`, `chips`, `junctions`, `qubits` or `test_strctures`), with path string starting from the [klayout_package](https://github.com/iqm-finland/KQCircuits/tree/main/klayout_package) or the [kqcircuits](https://github.com/iqm-finland/KQCircuits/tree/main/klayout_package/python/kqcircuits) folder.

The task is to come up with simple instructions to configure the Vim editor to perform this call on a file currently being viewed on Vim editor. This is probably best set up by adding a `:command` in `~/.vimrc`. Seems that using bang `!` to call KLayout is the best fit. If Vim's handling of file paths is incompatible with how it is implemented in [util/create_element_from_path.py](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py), feel free to change the code there. The ideal would be that no actual functionality is changed on that python file, but this can be negotiated in PR comments.

## Definition of done

- Find a streamlined way to configure Vim to call `klayout -e -rm util/create_element_from_path.py -rd element_path=...` from any OS platform
- Write up the instructions in the comment section of [util/create_element_from_path.py](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py) after the instructions for PyCharm and VSCode
- After following the instruction the user should be able to start a Vim session, open some python file from KQCircuits of some element for reading/writing, then typing `:kqc` to cause the KLayout window to open with the element shown as programmed in the opened python file

lavanya-m-k, 2023-06-01:
I would like to work on this issue. Can you please assign it to me?

qpavsmi, 2023-06-01:
> I would like to work on this issue. Can you please assign it to me?

Yes and thank you for your interest! You can start working on the solution and once the solution is ready, create a pull request. During review we might give some suggestions to refine the solution, but once it is ready, we will assign you to this issue and mark it closed. Don't hesitate to make a pull request earlier rather than later so that we could have some time before the deadline to reiterate your solution from our feedback if needed.

AVDiv, 2023-06-09:
Hey, 
I submitted a PR, Just check if any necessary changes to be done and notify me.

qpavsmi, 2023-06-10:
Just in case you havent seen it @AVDiv I added a review to your pull request, let me know if you cant see it

qpavsmi, 2023-06-10:
Marking this as done and confirming that @AVDiv should receive the bounty for this. The merging of the pull request will happen at a later date, but the code changes were checked and verified that they work! Well done!

closed 2023-06-10 by qpavsmi