iqm-finland/KQCircuits #69: HangerResonator breaks WaveguideComposite

iqmtestd, 2023-10-16:
### What happened? What you expected to happen?

WaveguideComposite Nodes may specify simple Elements to be embedded in the waveguide. This does not work well with HangerResonator. Often it breaks with a `ERROR: IndexError: list index out of range in PCellDeclaration.produce` message.

Other times  it does succeed but even then the guiding lines may be wrong:

[image]

The macro to create the above example:
```
from kqcircuits.klayout_view import KLayoutView
from kqcircuits.elements.waveguide_composite import Node, WaveguideComposite
from kqcircuits.elements.hanger_resonator import HangerResonator

view = KLayoutView()

nodes = [
    Node(pya.DPoint(0, 0)),
    Node(pya.DPoint(500, 0)),
    Node(pya.DPoint(500, -300)),
    Node(pya.DPoint(200, -300), face_id="2b1"),
    Node(pya.DPoint(0, -300)),
    Node(pya.DPoint(-200, -300)),
    Node(pya.DPoint(-200, -600), face_id="1t1"),
    Node(pya.DPoint(-200, -1000)),
    Node(pya.DPoint(-200,-1600), HangerResonator),
    Node(pya.DPoint(-500, -1000)),
]
view.insert_cell(WaveguideComposite, nodes=nodes)

view.focus()
```

Similarly, when trying to use `port_pl_a` and `port_pl_b`, like this:
```
    Node(pya.DPoint(-200, -1600), HangerResonator, align=('port_pl_b', 'port_pl_a')),
    Node(pya.DPoint(-500, -2000)),
```

it does not always produce the expected shape:
[image]
 

### Versions

KQCircuits: v4.7.0
KLayout: 0.28.12
Python: 3.10.12
OS: Ubuntu 22.04.3

### Issue Severity

Medium: significant difficulty but I can work around it

iqmtestd, 2023-10-16:
We don't know why this happens. A possible reason is that HangerResonator *uses* WaveguideComposite and the recursive nature of this usage may confuse KLayout. Or we just have some logic error in WaveguideComposite that has not been discovered yet.

closed 2023-10-20 by iqmtestd

Qvadi, 2023-10-24:
Awesome! Tested and works.

commit 45db7035 (2023-10-20): Fix HangerResonator (#70)

It uses the more efficient WaveguideCoplanar and thus also avoids recursive
element usage (disallowed by KLayout) when insertend in WaveguideComposite.

Fix #69