iqm-finland/KQCircuits #71: GitHub shuts down SVN bridge: KLayout package will need mirror at SourceForge

klayoutmatthias, 2023-10-22:
### Description

Starting Jan. 8 2024, GitHub SVN access will be disabled. This means, the convenient way of deploying KLayout packages will stop working. A short-term solution is to mirror the KLayout Package part to a Subversion project on SourceForge.

This topic is tracked here: https://github.com/KLayout/klayout/issues/1514

qpavsmi, 2023-10-23:
Thanks for letting us know, and thank you also for taking the time to migrate our current date KQC SALT package in your own SVN host!

Our SALT package release lifecycle is quite long, averaging about half a year. Since we made a SALT package release a couple of weeks ago, I believe by the time we prepare a new SALT package release, KLayout will already have git support for package manager implemented by that time.

Will keep this issue open until KLayout supports git for package management, or until some other development is made

klayoutmatthias, 2023-10-24:
Very good. Thanks for letting me know.

Matthias

qpavsmi, 2024-06-16:
After a bit of trial-and-error when releasing version `4.8.x`, I now have understanding how to make a successful Salt package release moving forward such that it uses GIT protocol. Closing issue

closed 2024-06-16 by qpavsmi