iqm-finland/KQCircuits #89: Reload Libraries macro issues: doesn't restore PCells on error, and fails for non-KQC PCells

qpavsmi, 2024-05-02:
### What happened? What you expected to happen?

I place a KQC element, for example the Double Pads from the Qubit Library. Then I open `klayout_package/python/kqcircuits/qubits/double_pads.py` and make `_build_island2` return empty `pya.Region()`. I go back to the same KLayout session and select `KQCircuits > Reload Libraries`. I get what I expected, which is that the bottom island on the qubit is not present anymore and there are no errors in `File > Log Viewer`.

Now I open `Basic` from the library panel, and place a Circle. For better demonstration I double click the Circle cell, choose `PCell Parameters` and change `Radius` to 100 microns and `layer` to `1t1 base metal wo grid 130/1`. Now do some change in double pads code (e.g. revert change made above) and try `KQCircuits > Reload Libraries`. Now it gives error `AttributeError: type object 'PCellDeclaration_Native' has no attribute 'get_schema'`. Also the double pads qubit will disappear.

Will provide more information later.

### Versions

Version independent issue

### Issue Severity

Low: it is annoying

PietroCampana, 2024-06-04:
Hi, I would like to work on this issue as part of unitaryhack 2024

qpavsmi, 2024-06-04:
> Hi, I would like to work on this issue as part of unitaryhack 2024

Sounds great! I encourage to make a pull request early even if the solution is not completely finalized to get early feedback.

qpavsmi, 2024-06-10:
Completed by https://github.com/iqm-finland/KQCircuits/pull/95 thanks @PietroCampana

closed 2024-06-10 by qpavsmi