iqm-finland/KQCircuits #92: Broken `Edit on Github` link for API pages of KQC documentation

qpavsmi, 2024-05-02:
### What happened? What you expected to happen?

I open an API page of KQC class, for example https://iqm-finland.github.io/KQCircuits/api/kqcircuits.elements.smooth_capacitor.html

I click the `Edit on Github` link and it attempts to redirect me to `https://github.com/iqm-finland/KQCircuits/blob/main/docs/api/kqcircuits.elements.smooth_capacitor.rst`, which does not exist. This happens for any page under the `API documentation` section.

Reason is that we configured [sphinx](https://www.sphinx-doc.org/en/master/) to generate docs for all KQC classes and modules. These pages are autogenerated, and don't have a documentation "source" the same way hand-written documentation pages have.

KQC documentation generation should be reconfigured so that for API pages it doesn't show the `Edit on Github` link. This post that should help to get started: https://github.com/sphinx-doc/sphinx/issues/2386#issuecomment-478403897

### Versions

Version independent

### Issue Severity

Low: it is annoying

f-martini, 2024-05-30:
Hi! I would like to try to solve this issue as part of unitaryHACK 2024.

qpavsmi, 2024-05-30:
Thanks! Feel free to post here or use unitaryhack discord `#kqcircuits` channel if you need additional support!

f-martini, 2024-05-30:
> Thanks! Feel free to post here or use unitaryhack discord `#kqcircuits` channel if you need additional support!

Thanks to the link you provided above, I was able to solve the issue. 
I've submitted a QM Individual Contributor License Agreement as specified in the [guidelines](https://github.com/iqm-finland/KQCircuits/blob/b3082b0e1fd503facb04a663e1e43e5cd26e206f/docs/contributing.rst), is there anything else I need to do to send a pull request? (currently I'm working on a fork of the project).
Also, is there any template I can use to properly submit the pull request?
Thanks in advance! :)

qpavsmi, 2024-05-30:
You can submit the pull request even now if you want. The cla-bot only prevents the pull request from merging, but before that I still need to review the changes. We don't have a template, just push your branch and make a pull request out of it.

PR #93 (2024-05-31): Removal of 'Edit on Github' link for API pages of KQC documentation - Fixes #92
### ISSUE
The API documentation pages are automatically generated by sphinx. Despite not having a documentation "source" the same way hand-written documentation pages have, sphinx still renders a link to the associated non-existent repo page.  
Such link has to be removed only for the API-related documentation pages.

### SOLUTION

Added  docs/_templates/breadcrumbs.html, as suggested in [this post](https://github.com/sphinx-doc/sphinx/issues/2386#issuecomment-478403897), in order to overwrite the custom sphinx the [... cut]
commit 08d7d8a9 (2024-05-31): Removal of 'Edit on Github' link for API pages of KQC documentation - Fixes #92 (#93)

* added breadcrumbs.html to manage 'github_url' meta

* added github_url meta to relevant files preventing undesired github link rendering
closed 2024-05-31 by qpavsmi