iqm-finland/KQCircuits #100: Ansys export issue

SimonTheRocketGuy, 2024-07-11:
### What happened? What you expected to happen?

[image]
Cant export to ansys always throws error even after editing code

### Versions

KQcircuits 4.8.2 Klayout0.29.3 WIN10

### Issue Severity

High: blocks me from doing my work

qpavsmi, 2024-07-11:
Hi! Is this reproducible by running one of the scripts under `klayout_package/python/scripts/simulations`?

SimonTheRocketGuy, 2024-07-11:
[image]
running example create capacitive pi model

qpavsmi, 2024-07-11:
At least scripts under `klayout_package/python/scripts/simulations/ansys`  like `klayout_package/python/scripts/simulations/ansys/create_capacitive_pi_model.py` are not intended to be run directly by user, but by other scripts or by the Ansys executable.

I noticed that the exception is being caught in the KLayout window, are these attempted to be run by KLayout Macro editor? Instead of running the scripts as a macro, we currently recommend using a terminal or command prompt to run `kqc sim <script name>`, instructions here: https://iqm-finland.github.io/KQCircuits/user_guide/simulation/export_and_run.html

You can use `kqc sim ls` to see which simulation scripts are available for running.

Before using `kqc sim`, make sure that KQC is installed in standalone following this guide: https://iqm-finland.github.io/KQCircuits/developer/standalone.html#basic-installation

And make sure `simulations` is included as a dependency extension.

qpavsmi, 2024-07-16:
Hi @SimonTheRocketGuy , just wanted to follow up if you are still experiencing problems with exporting simulations to Ansys?

SimonTheRocketGuy, 2024-08-20:
Hello, I tried everything. But I was unable to fix it. Maybe it is bug. Can we connect on linkedin and chat a bit please?

qpavsmi, 2024-08-20:
Sure, my LinkedIn https://www.linkedin.com/in/pavel-smirnov-a7597a151/

Or if you have discord: _pavelsmirnov

SimonTheRocketGuy, 2024-08-20:
Already send connection request