iqm-finland/KQCircuits #101: Type error in Meander object using last release of KLayout in windows

jevillegasd, 2024-08-30:
### What happened? What you expected to happen?

Hey! When instantiating a meander PCell, an error **'TypeError: 'DPoint' is not callable** is thrown. The error source points to the PCell declaration helper function, so I couldn't trace which line in the meander definition is generating the error.  Other PCells don't seem to throw this error. I tried this using main and the latest salt package direc;ty installed from KLayout and I get the same error.

[image]

Best,
Juan

### Versions

KQCircuits: 4.8.2 (and main at commit #[90c2c70](https://github.com/iqm-finland/KQCircuits/commit/90c2c704f7fa06cc203c1f678aff2c139a9b70e1))
KLayout: 0.29.6 
OS: Windows 11

### Issue Severity

Medium: significant difficulty but I can work around it

qpavsmi, 2024-08-30:
Hi! This is an issue I am working on right this moment. Starting from KLayout 0.29.5 the `PCellDeclarationHelper` class from which all `Element` subclasses are derived got a `start` function introduced. This caused `Meander` element to fail (maybe others too) because it has a parameter named `start`, that is currently clashing with KLayout's code. Expect a fix for this early next week or maybe even today.

qpavsmi, 2024-08-30:
But yes, in the mean time, the workarounds are:
- Downgrade KLayout to 0.29.4 or older
- Rename `start` parameter in Meander, also make sure that any chips you want to use also reference the renamed parameter

qpavsmi, 2024-09-04:
This has now been fixed as of KQCircuits v4.8.4, its compatible with KLayout >= 0.29.5. Closing issue. Feel free to reopen the issue if something is still wrong.

closed 2024-09-04 by qpavsmi