iqm-finland/KQCircuits #103: Attempting to use Developer Notebook

M3K33L, 2024-09-30:
### What happened? What you expected to happen?

I am trying to use the example notebook from the developer setup to create a new element (to avoid reloading my library to check my progress over and over). I opened the notebook by using Anaconda Navigator's installed Jupyter Lab, within my virtual environment for KQC. The first time I used the notebook, it worked fine, but the second time I ran the kernel, I got this error:
```
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[3], line 1
----> 1 view.insert_cell(Demo, name_chip="ABC")

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\klayout_view.py:134](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/klayout_view.py#line=133), in KLayoutView.insert_cell(self, cell, trans, inst_name, label_trans, align_to, align, rec_levels, **parameters)
    106 def insert_cell(
    107     self, cell, trans=None, inst_name=None, label_trans=None, align_to=None, align=None, rec_levels=0, **parameters
    108 ):
    109     """Inserts a subcell into the first top cell (the very first cell in the cell window)
    110 
    111     It will use the given ``cell`` object or if ``cell`` is an Element class' name then directly
   (...)
    131         tuple of placed cell instance and reference points with the same transformation
    132     """
--> 134     return insert_cell_into(
    135         self.top_cell, cell, trans, inst_name, label_trans, align_to, align, rec_levels, **parameters
    136     )

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\elements\element.py:87](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/elements/element.py#line=86), in insert_cell_into(target_cell, cell, trans, inst_name, label_trans, align_to, align, rec_levels, **parameters)
     85 layout = target_cell.layout()
     86 if isclass(cell):
---> 87     cell = cell.create(layout, **parameters)
     89 if trans is None:
     90     trans = pya.DTrans()

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\elements\element.py:197](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/elements/element.py#line=196), in Element.create(cls, layout, library, **parameters)
    188 @classmethod
    189 def create(cls, layout, library=None, **parameters) -> pya.Cell:
    190     """Create cell for this element in layout.
    191 
    192     Args:
   (...)
    195         **parameters: PCell parameters for the element as keyword arguments
    196     """
--> 197     cell = Element._create_cell(cls, layout, library, **parameters)
    198     setattr(cell, "length", lambda: get_cell_path_length(cell))
    199     return cell

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\elements\element.py:494](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/elements/element.py#line=493), in Element._create_cell(elem_cls, layout, library, **parameters)
    492     return layout.create_cell(cell_library_name, parameters)
    493 else:
--> 494     load_libraries(path=elem_cls.LIBRARY_PATH)
    495     return layout.create_cell(cell_library_name, elem_cls.LIBRARY_NAME, parameters)

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\util\library_helper.py:88](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/util/library_helper.py#line=87), in load_libraries(flush, path)
     77 """Load all KQCircuits libraries from the given path.
     78 
     79 Args:
   (...)
     85      A dictionary of libraries that have been loaded, keys are library names and values are libraries.
     86 """
     87 # Try reloading all pcells before deleting libraries, otherwise classes are lost in case of errors
---> 88 all_pcell_classes = _get_all_pcell_classes(flush, path)
     90 if flush:
     91     delete_all_libraries()

File [~\Documents\KQCircuits\klayout_package\python\kqcircuits\util\library_helper.py:306](http://localhost:8890/~/Documents/KQCircuits/klayout_package/python/kqcircuits/util/library_helper.py#line=305), in _get_all_pcell_classes(reload, path, skip_modules)
    303 import_path_parts = mp.parts[::-1][mp.parts[::-1].index(pkg) :: -1]
    304 import_path = ".".join(import_path_parts)[:-3]  # the -3 is for removing ".py" from the path
--> 306 module = importlib.import_module(import_path)
    307 if reload:
    308     importlib.reload(module)

File [~\.conda\envs\KQCandEPR\Lib\importlib\__init__.py:126](http://localhost:8890/~/.conda/envs/KQCandEPR/Lib/importlib/__init__.py#line=125), in import_module(name, package)
    124             break
    125         level += 1
--> 126 return _bootstrap._gcd_import(name[level:], package, level)

File <frozen importlib._bootstrap>:1204, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1176, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:1126, in _find_and_load_unlocked(name, import_)

File <frozen importlib._bootstrap>:241, in _call_with_frames_removed(f, *args, **kwds)

File <frozen importlib._bootstrap>:1204, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1176, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:1126, in _find_and_load_unlocked(name, import_)

File <frozen importlib._bootstrap>:241, in _call_with_frames_removed(f, *args, **kwds)

File <frozen importlib._bootstrap>:1204, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1176, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:1140, in _find_and_load_unlocked(name, import_)

ModuleNotFoundError: No module named 'kqcircuits.chips.'
```

I uninstalled the Salt package, and uninstalled KQC. Then, with a fresh git pull, I reinstalled KQC developer with
```
python -m pip install -e "klayout_package/python[docs,tests,notebooks,simulations,graphs]"
```
Which resulted in the following:
```
Successfully built kqcircuits
Installing collected packages: kqcircuits
Successfully installed kqcircuits-4.8.2.post5+git.5ace7e68.dirty
```
Then, I tried to go back to the notebook, but still got the same error. I'm not sure where to go next.

### Versions

KQCircuits: 4.8.2
KLayout: 0.29.4
Python: 3.11.9
OS: Windows Version 10.0.22631 Build 22631

### Issue Severity

Medium: significant difficulty but I can work around it

qpavsmi, 2024-10-02:
Hi! Do you rerun just the one cell, or the entire notebook from the beginning? Latter is preferred.

Could you try running the notebook code in the python interpreter, to see if that causes errors too? If you just type `python` in command line, it should open an interactive console where you can write python code. You can always write `exit()` to close it.

Then in python interpreter write these code lines one by one and let me know if there are errors:
```
from kqcircuits.klayout_view import KLayoutView
from kqcircuits.chips.demo import Demo
view = KLayoutView()
view.insert_cell(Demo, name_chip="ABC")
```

M3K33L, 2024-10-02:
Hi! I ended up deleting all files related to KQC from my computer, then doing a new git clone. Then, I made a venv through python rather than conda. Using that venv, I pip installed KQC from scratch as a standalone. Then, I opened Spyder to test out the view, and at first it said that no module kqcircuits existed. To fix that, I restarted kernel and found that I needed to pip install spyder-kernel in my venv. Then, it worked! Success!

I do have some questions though. The way that I was trying to debug my chip was to include a new variable
```
testrefs = self.refpoints
print(testrefs)
```
Which let me view the refpoints as I added elements to my chip. Is there a better way that you'd recommend?

Also, for general development of an element, should I create a class in this program then add that to the view? Or should I create my element elsewhere and re-run my code after making edits in the other file? The reason I ask is: whenever I modified my chip, I had to restart kernel for the changes to take effect. I haven't gotten to try modifying a class within this same program.

qpavsmi, 2024-10-08:
Recommended way for now is to include the code where you implement your element in the directories of the source code, that is chips in [klayout_package/python/kqcircuits/chips](https://github.com/iqm-finland/KQCircuits/tree/main/klayout_package/python/kqcircuits/chips), elements in [klayout_package/python/kqcircuits/elements](https://github.com/iqm-finland/KQCircuits/tree/main/klayout_package/python/kqcircuits/elements), junctions in [klayout_package/python/kqcircuits/junctions](https://github.com/iqm-finland/KQCircuits/tree/main/klayout_package/python/kqcircuits/junctions) etc. If you re-run the cell where you import your element, it should then get the most recent version of your element, I don't think reloading the kernel should be necessary.

If you're using jupyter notebooks then if you put the following code in the cell, then rerun the import cell and then the following cell, it should show you visually how your element looks like in its most up-to-date version:

```
view.insert_cell(YourElement)
view.show(width=800)
```

Using jupyter notebooks might be a bit constrained as it only allows you to interact with your design through pictures. We would recommend installing KLayout as a GUI application [as instructed here](https://iqm-finland.github.io/KQCircuits/developer/setup.html). Then you can find your design in the bottom left "Libraries" panel and you can drag-and-drop your design into the layout. Then you can double-click your design and change the PCell parameters to see that they work as intended. More instructions [here](https://iqm-finland.github.io/KQCircuits/user_guide/gui_features.html).

If you use pycharm or Visual studio code editors to write your code, you can set up a shortcut such that it takes the code file you have open, then boots KLayout with that element centered and zoomed out, which you can do during the iterative process of refining your design. Instructions on how to set that up is [here](https://github.com/iqm-finland/KQCircuits/blob/main/util/create_element_from_path.py)