iqm-finland/KQCircuits #108: Improve visualisation of simulation results with Paraview

qpavsmi, 2025-05-09:
## Prerequisites

1. KQCircuits [standalone](https://iqm-finland.github.io/KQCircuits/developer/standalone.html)
  - Install core, ``tests`` and ``simulations`` requirements
  - We recommend using [venv](https://docs.python.org/3/library/venv.html) to contain requirements
2. KQCircuits [GUI installation](https://iqm-finland.github.io/KQCircuits/developer/setup.html)
3. [Elmer FEM simulator](https://www.elmerfem.org/blog/binaries/)
4. [Paraview](https://www.paraview.org/) to visualize simulation results

Points 2. and 3. can be skipped for the purposes of getting this issue done, if there is trouble getting these prerequisites working.

## Description

After running `kqc sim ...` command, the [paraview](https://www.paraview.org/) application is invoked to visualise simulation results. Paraview is invoked to simply load the result file, but there are still manual steps to do to actually visualise simulated element. Using paraview's [scripting capabilities](https://docs.paraview.org/en/latest/Tutorials/SelfDirectedTutorial/batchPythonScripting.html), we can automate this part.

Paraview tool is only relevant for Elmer simulations done with KQCircuits. Ansys simulations are integrated into the Ansys tool, so visualization is also handled by Ansys.

## Introduction to simulation visualization with Paraview

For the purposes of this issue, simulations that are done with KQCircuits are divided into two categories: 3D simulations and 2D simulations.

### 3D simulations
Let's run an example 3D simulation. Open [concentrictransmon_couplings_sim.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/simulations/concentrictransmon_couplings_sim.py) for editing. Let's make some changes to the script for demonstration purposes.

In `elmer_export_parameters` dictionary, in `workflow`, change `run_paraview` value to True. This enables prompting Paraview to visualize results.

If you had trouble with prerequisite point 2, replace the `open_with_klayout_or_default_application(export_simulation_oas(simulations, dir_path))` call with `export_simulation_oas(simulations, dir_path)`. The script invokes KLayout GUI to preview geometry before simulation, which is not strictly needed to develop this feature.

With these changes made, open terminal and run

`kqc sim concentrictransmon_couplings_sim.py`

The script will sweep 16 qubit geometries, simulate each sweep one at a time, then open a Paraview window. Closing the window will proceed to simulating next sweep.

[image]

Let's keep a Paraview window open and see what manual steps need to be taken to properly visualize the element. First, click on the `Apply` button in left-hand bottom `Properties` panel. In left-hand `Pipeline Browser` panel, there are three files:
- `concentrictransmon_*_t0001.pvtu` - inner island junction port excited
- `concentrictransmon_*_t0002.pvtu` - outer island junction port excited
- `concentrictransmon_*_t0003.pvtu` - coupler port excited

[image]

[image]

Click one of the files, then click the `Threshold` tool from above toolbar. In below `Properties` panel, set `Scalars` to `GeometryIds`, then set `Lower Threshold` and `Upper Threshold` to `1` (pay attention, Paraview will sometimes sneakily set a floating point number close to 1, then trim the number. The value has to be exactly `1`). Then click the `Apply` button again. You should now see a colormap of the excitation. The view is still zoomed out a bit, use the mouse scrollwheel to zoom in the simulation box. If the box is off center, press down scroll wheel and drag the mouse to laterally move the simulation box. If you accidentally move the view in unwanted direction and its difficult to recenter the view, use the `Set view direction to -Z` button.

You can `Ctrl+C` in terminal if you wish to abort simulation execution without processing all sweeps.

Let's try another example. Open [circular_capacitor_epr_sim.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/simulations/circular_capacitor_epr_sim.py) for editing. Set `"run_paraview": True` like in previous example. Then relax the simulation mesh to speed up simulation time by sacrificing fidelity, within `ElmerEPR3DSolution` change `mesh_size` value to:
```
{
        "global_max": 80.0,
        "1t1_gap&1t1_signal_1": [4.0, 8.0],
        "1t1_gap&1t1_signal_2": [4.0, 8.0],
        "1t1_gap&1t1_ground": [4.0, 8.0],
        "2b1_gap&2b1_ground": [4.0, 8.0],
}
```
Then, to be safe, change `workflow.update` call so that `n_workers` is set to `1`. Some multiprocess features were found not to work on certain systems. Now run the script. Do notice that to simulate a single sweep it takes a longer time for this simulation, 15 minutes on standard laptop.

`kqc sim circular_capacitor_epr_sim.py`

Notice that the number of `.pvtu` files is different. Different simulations may have different number of ports.

Repeat the manual steps on paraview window. As you set the upper and lower threshold, you will notice that setting them to `1` will not cause intended result. The correct values for this simulation are lower threshold `4` and upper threshold `20`.

Where do these values come from? Open the folder where you cloned `KQCircuits` to. There should now be `tmp` folder, and in it `concentrictransmon_couplings_sim_output` and `circular_capacitor_epr_sim_output` subfolders, each created during your previous simulation runs. Look inside the `circular_capacitor_epr_sim_output/circular_capacitor_epr_30` folder. You should see the two familiar `*.pvtu` files there, these are in-fact opened using the following command during preview:

`paraview circular_capacitor_epr_30_t0001.pvtu circular_capacitor_epr_30_t0002.pvtu`
[... 60 lines cut ...]
4. `simulation.sh` executes each `*.sh` script of each sweep
5. A `*.sh` script repeatedly calls copied-over [run.py](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/simulations/elmer/run.py) script to perform various steps of executing elmer simulation: meshing, running Elmer, then **visualizing with Paraview**.
6. With `--only-paraview` option, the `run.py` calls [run_paraview](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/scripts/simulations/elmer/run_helpers.py) function.
7. `run_paraview` currently calls `paraview` command on the `*.pvtu/*.vtu` files.

So we should modify point 7 so that `paraview` command is given a python file instead of `*.pvtu/*.vtu` files as argument. `paraview` will run that python script in its own python environment, and unfortunately it has a limitation where you cannot pass arguments to the python script (as far as we know, we'd be delighted if you are able to find some support for this on Paraview-side).

At point 2, all python files under `klayout_package/python/scripts/simulations/elmer` are copied to `tmp/<simulation_output>/scripts`, which would include `paraview/center_3d.py` you implemented. The recorded routine in `center_3d.py` could be encapsulated in a function called e.g. `run()`, which accepts arguments, e.g. file path to `*.pvtu` file.

At point 2 for each sweep, in addition to `*.gds`, `*.json` file, we should also generate a small python file, let's call it `*_pv.py` file (for example `circular_capacitor_epr_30_pv.py`). Contents of this file could look like:
```python
import sys
sys.path.insert(0, <PATH_TO_SCRIPTS_FOLDER>)
from center_3d import run
run(<PVTU_FILES>, <OTHER_NEEDED_ARGUMENTS>)
```
where we could substitute `<PVTU_FILES>` and `<OTHER_NEEDED_ARGUMENTS>` for each sweep specifically. Now at point 7 the `run_paraview` function should pass the python file of the sweep to paraview executable e.g. `paraview circular_capacitor_epr_30_pv.py`.

The recorded macro for 2D simulations is probably different from 3D, and probably should be a separate generic python file. Best way to decide if the simulation is a 2D simulation, is in [export_elmer](https://github.com/iqm-finland/KQCircuits/blob/main/klayout_package/python/kqcircuits/simulations/export/elmer/elmer_export.py) function check if `solution` `tool` attribute is `"cross-section"`, in which case it is a 2D simulation. In all other cases best to assume it is a 3D simulation.

With these changes in place, see if setting `run_paraview` to True and running `kqc sim concentrictransmon_couplings_sim.py` will prompt Paraview with visualisation working off-the-bat!

## Definition of done
- After running `kqc sim ...`, the user should eventually see a Paraview window open with good visualization of simulation result without manual input.
- No unit tests or documentation update needed

jbolns, 2025-05-28:
I subscribed to this issue in case it is still available after UnitaryHack. Looks interesting.

qpavsmi, 2025-05-31:
> Hello team! I want to try my hands at creating solution for this issue, as part of unitaryHack 2025!

Thank you for your interest! Looking forward to your pull request! Feel free to submit one even if it's not fully complete for a chance to receive early feedback.

Also, a reminder for other unitaryHack participants reading this, you are also welcome to submit your solution to this issue.

jbolns, 2025-06-05:
@qpavsmi 

I started playing around with this today.

Main reason I'm not sure I'll get it done by unitaryHack deadlines is that I'm not familiar with the tools and there are many steps involved, each with room for having to figure out things.

But it's very interesting and worth doing even as a learning experience.

So, If someone swoops it from me, well, that's life. 

***That said,*** I managed to get things running on WSL, visualised some steps from the first example (the one below is from the second sweep, not necessarily part of the official workflow - was just playing around with it to familiarise with tool), and saved some test macros.

[image]

**I have some questions already**
1. Is the goal a single example for one simulation, like a proof of concept that can later be replicated simulation-at-a-time, or a generalisable script or script-family solution flexible enough to automate all simulations. 
2. Is there a way to reduce the computational requirements of ParaView? It tells me (bottom right corner) it is only using a small amount of resources, but it is sluggish, sometimes will freeze upon load, and other times won't actually render things.
3. What's the level of detail needed in the macros (or otherwise-coded scripts)? See screenshot below for the specific context for this question.

[image]

jbolns, 2025-06-06:
Thanks, @qpavsmi 

2 & 3. No need to put time on the question about paraview's computing demands. I was wondering if there are known easy hacks, but this is not a critical need. I will not be using the code saved from macros directly but doing exactly what you suggest in 3: run a few macros just to get a sense of the commands and then code things efficiently. From prev. experiences with software with the ability to record macros, code saved from macros tends to be incredibly bloated. Cleaning it is about as difficult as just understanding the logic and coding it anew. 

1. Right, so, first thing I'll try to get working is some kind of interconnector to pass settings from simulation outputs into Paraview upon spawning it. This seems to be the critical piece of the puzzle. Since it's critical, if I manage, I'll make a PR with the raw pipeline and a super minimal example.

qpavsmi, 2025-06-06:
@jbolns Sounds good! Check `Integrating recorded macro into KQC simulation pipeline` section for an idea I had in mind on how to pass data into the macro. Generally it involves parsing external files at expected file paths

jbolns, 2025-06-09:
I have pushed a semi-final version of my approach. Happy to adjust, of course, but considering the time limits associated with the unitaryHacck, I reckon the general approach and implementation are fairly set for the time being. 

I do NOT know if it will be better or worse than the competing version. Chances are @InderpalSuthar will push a comparable or better solution. Regardless, I do want to apologise publicly to him/her if he/she chose the issue partly because I noted I would work on it after unitaryHack. My initial assessment was not ill intended. It came out of lack of confidence on my skills.

Buy me a beer if you win. I will do the same if I do. 

@qpavsmi

qpavsmi, 2025-06-09:
@jbolns Thanks! I'll be reviewing both versions and will choose which will fit our needs best by hackathon deadline. I will try to provide good feedback before then as I go along to give you both the best shot.

closed 2025-06-12 by qpavsmi